TheThingsNetwork/arduino-device-lib #5: Simplify: Define Serials and their baud rate for known devices

FokkeZB, 2016-08-19:
The library obviously has to work with any Arduino device, but can we simply sketches for The Things Uno, Node and maybe other common devices?

For example, combined with #4 the following sketch:

```c
#define debugSerial Serial
#define loraSerial Serial1
// ..
#define debugPrintLn(...) { if (debugSerial) debugSerial.println(__VA_ARGS__); }
#define debugPrint(...) { if (debugSerial) debugSerial.print(__VA_ARGS__); }
// ..
debugSerial.begin(115200);
loraSerial.begin(57600);
// ..
ttn.init(loraSerial, debugSerial);
// ..
debugPrintLn("Setup for The Things Network complete");
```

Could be simplified to:

```c
ttn.initWith(TTN_UNO);
// ..
ttn.debugPrintLn("Setup for The Things Network complete");
```

FokkeZB, 2016-09-05:
Maybe `initFor` would be a better name.

@johanstokking am I right there are three "variables" the library needs to know to work on any Arduino:

* The type of LoRaWAN module used (which we internally [map](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L120) to settings)
* The Serial Port the module is on
* The Serial Port used for debug

Since we can [detect the module](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L97-L104) and `Serial` is [always for USB](https://www.arduino.cc/en/Reference/Serial) we only really need to know the Serial Port the LoRaWAN module is on right?

If I'm right then I think we could just remove the `init()` function and add methods to override the Serial Port for the LoRaWAN module in case you don't have it on `Serial1` (or would like to use a different baud rate)

```c
ttn.setModemSerial(modemSerial);
```

This and removing `reset()` will simplify a lot:

```c
#include <TheThingsNetwork.h>

const byte appEui[8] = {0x70, 0xB3, 0xD5, 0x7E, 0xE0, 0xE0, 0x01, 0x4A1};
const byte appKey[16] = {0x73, 0x6D, 0x24, 0xD2, 0x69, 0xBE, 0xE3, 0xAE, 0x0E, 0xCE, 0xF0, 0xBB, 0x6C, 0xA4, 0xBA, 0xFE};

TheThingsNetwork ttn;

void setup() {
  while(!ttn.join(appEui, appKey));

  ttn.showStatus();
}

void loop() {
  ttn.sendBytes(payload, sizeof(payload));
  
  delay(20000);
}
```

johanstokking, 2016-09-06:
The challenge here is that the entire `Serial1` keyword, regardless of initialization, is not known on for example the Arduino Uno board. The only way to use it, is either by passing it from the user sketch, or a `#define` before including the header, so that the header can conditionally reference `Serial1`.

FokkeZB, 2016-09-06:
But we can test if `Serial1` exists right? So if it does we can use that (which would work for our Uno and Node). Same goes for `Serial` for debug. And then we'd have `setDebugSerial()` and `setModemSerial()` to override. We could postpone calling `.begin()` to `reset()` (which is called from both `personalize()` and `join()` now) so the user has every change to override before we actual use them.

johanstokking, 2016-09-06:
Yes, but I don't know if it's good to assume that if `Serial1` is defined, then that is automatically the modem serial. But indeed, we can have the overrides.

FokkeZB, 2016-09-06:
I think it's fine to optimise (minimise) the library for our Uno and Node and have overrides for other devices if needed.

johanstokking, 2016-09-19:
There are too many uncertainties and it may cause more issues than that it solves a problem (if any)

closed 2016-09-19 by johanstokking