TheThingsNetwork/arduino-device-lib #6: Simplify: Automatically reset() before join() and personalize()

FokkeZB, 2016-08-19:
All examples call `ttn.reset()` before they call `ttn.join()` or `ttn.personalize()`. If this is a required step, why not do this within the library so that we can simplify sketches from:

```c
ttn.init(loraSerial, debugSerial);
ttn.reset();
ttn.personalize(devAddr, nwkSKey, appSKey);
```

to:

```c
ttn.init(loraSerial, debugSerial);
ttn.personalize(devAddr, nwkSKey, appSKey);
```

johanstokking, 2016-08-20:
Yes, I think we can do the `reset()` in `personalize()` and `join()`. The reason why it's separate, was to be able not to reset and join with the values that are stored in the RN module, instead of providing the values. But we can account for that in `join()` and/or with an overload that doesn't take keys and therefore doesn't reset.

FokkeZB, 2016-08-22:
Yes, that seems like a good simplification since doing a `reset()` or `personalize()` without `reset()` will be the common scenario. I'll leave this assigned to you as I'm finding my way around in C/Arduino still ;)

FokkeZB, 2016-09-06:
Of course if we call `reset()` from `join()` this will mean that if it takes a while to do a successful join, every new attempt will first `reset()` the device. This is a bit of an overhead.

johanstokking, 2016-09-06:
Also, I think we should make it optional not to reset the device at all. I can imagine the AppEUI and AppKey being pre-flashed, especially in The Things Node, that we don't want to reset.

FokkeZB, 2016-09-22:
Closed by #58

closed 2016-09-22 by FokkeZB