TheThingsNetwork/arduino-device-lib #7: Simplify: Automatically delay where needed

FokkeZB, 2016-08-19:
All examples use `delay()` with various values between different calls to other `ttn` methods. If I remember correctly from the TNW Hack Battle leaving this out could break the sketch.

If these delays are required, why not handle this in the library so that sketches are more cleaner and simple?

Before:

```c
void setup() {
  // ..

  delay(1000);

  ttn.init(loraSerial, debugSerial);
  ttn.reset();
  ttn.personalize(devAddr, nwkSKey, appSKey);

  delay(6000);

  ttn.showStatus();

  debugPrintLn("Setup for The Things Network complete");

  delay(1000);
}
```

After:

```c
void setup() {
  // ..

  ttn.init(loraSerial, debugSerial);
  ttn.reset();
  ttn.personalize(devAddr, nwkSKey, appSKey);
  ttn.showStatus();

  debugPrintLn("Setup for The Things Network complete");
}
```

johanstokking, 2016-08-20:
They should not be necessary. It's in the example sketches for the user to have time to connect the serial monitor to see the status, in this case.

Let's try to remove the delays and see where we have problems. If so, then we indeed need to put the delay there.

FokkeZB, 2016-08-22:
I'll remove them and test.

closed 2016-08-22 by FokkeZB

FokkeZB, 2016-08-22:
We can always suggest starting the setup with the following:

```
while(!debugSerial) { sleep(1000) }
```

johanstokking, 2016-08-22:
@FokkeZB that's often used in examples, but it's a bad practice. If you then try with a power bank (which doesn't open debug serial), it doesn't work and you'll won't see why. We can use it, but then with a time-out, or a `#define DEBUG` that people should not forget to enable/disable.

FokkeZB, 2016-08-23:
True

Maybe if we provide #4 we could buffer the last X nr of debugPrint calls and once serial is connected spit these out.