TheThingsNetwork/arduino-device-lib #11: sendBytes return value

waltertakens, 2016-08-23:
now sendBytes returns an integer containing the number of bytes received downlinkLength,
 or in case of unconfirmed 0.

Is it easier for the enduser to still return an integer but composed of 2 parts :
msb signed byte  containing the result of the send , eg 0 for ok, -1 for timeout, -2 for busy etc
lsb unsigned byte containing the received downlinkLength

FokkeZB, 2016-08-23:
@johanstokking can we maybe simplify downlink some while we're at it? I find it weird that you sendBytes in order to receive. I know this is how it technically works, but it's not very intuitive.

**current**

```c
void loop() {
  // Send a byte
  byte buf[1];
  buf[0] = 20;
  int downlinkBytes = ttn.sendBytes(buf, 1);

  if (downlinkBytes > 0) {
    debugPrintLn("Received " + String(downlinkBytes) + " bytes")
    // Print the received bytes
    for (int i = 0; i < downlinkBytes; i++) {
      debugPrint(String(ttn.downlink[i]) + " ");
    }
    debugPrintLn();
  }

  delay(20000);
}
```

**possible simplification**

```c

// let users define a function to pass as callback
void receive(message) {
  // .. do something
}

void loop() {

  // new function handling sendBytes and then looping downlink
  ttn.pingDownlink(receive);

  delay(20000);
}
```

I'm no C/Arduino expert so I don't know if there are limitations that make this impossible or maybe we can even register the callback and an interval in `setup()`? Or is it a single thread?

cambierr, 2016-08-24:
@FokkeZB It's how it often works in public libs (see Wire lib for instance). User registers a callback function which is then called by the lib when necessary.

FokkeZB, 2016-09-09:
I've deleted the discussion on an alternative API for receiving messages. Let's do that in #23.

For `sendBytes`, we still need a better response. Currently it returns the number of bytes received in response, but we won't need that anymore with #23.

Instead, it should return a success or error code, e.g.:

- `1` if the [4th `confirm` argument](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L250) was set to to `true` and the message was confirmed indeed.
- `0` if the [4th `confirm` argument](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L250) was not set to `true` and there were no errors.
- `-1` if [sendCommand](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L252) failed.
- `-2` if there was a [timeout](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L259)
- `-3` if there was an [unexpected response](https://github.com/TheThingsNetwork/arduino-library/blob/master/src/TheThingsNetwork.cpp#L279)

waltertakens, 2016-09-10:
What about extending the send returncodes with
0  if the 4th confirm argument was set to to true and there was no confirmation within the timeout limit

Furthermore, in case of an error you can react in your sketch in various ways
1) debug your sketch :-) , configuration params, and/or hardware
2) do nothing, hope that a future send will succeed. If that one also fails goto either 1) or 3)
3) reset the module (automatically in the sketch), rejoin, retry
4) explicitly schedule a new send later (eg in case of "busy" or "no_free_channel")
5) ....

In the proposed errorcodes there is not much granularity, to decide which is the best reaction
I can imagine that you dont want to expose the full rn module error list, you want to shield complexity for a beginner. However the best way to help that beginner is to propose how the fixing of the error is to be addressed.

FokkeZB, 2016-09-12:
@waltertakens I've updated my list to include that option. I've used truthy error codes for confirmed/no-error and falsy for error as well as the case that confirmation was asked but not received.

The docs will have these codes documented with suggestions, but we might also write debug logs whenever one happens with the suggestion.

FokkeZB, 2016-09-22:
Closed by #37

closed 2016-09-22 by FokkeZB

waltertakens, 2016-09-22:
Great ! 
Please also update the interface doc https://www.thethingsnetwork.org/docs/arduino/

FokkeZB, 2016-09-22:
Will be closed by #57