TheThingsNetwork/arduino-device-lib #14: Simply: sendBytes

FokkeZB, 2016-09-05:
Current:

```c
int sendBytes(const byte* buffer, int length, int port = 1, bool confirm = false);
```

Simplified:

```c
int sendBytes(const byte* buffer, int port = 1, bool confirm = false);
int sendBytes(const byte* buffer, bool confirm = false);
```

Reasoning:

* Examples have `ttn.sendBytes(payload, sizeof(payload))` so why not do `sizeof(payload)` in the lib and go from 4 to 3 args.
* With an additional version without port you can ask for confirm without having to set port.

Nicolasdejean, 2016-09-06:
If I remove port in sendBytes() I can also remove it in sendString(), right?

johanstokking, 2016-09-06:
@FokkeZB you cannot do `sizeof()` of a pointer. The size is known in the sketch, because it's declared as `byte buffer[4]` for example.

@Nicolasdejean yes those should be consistent. But I think that `port` should always be part of the `send...()` method.

FokkeZB, 2016-09-06:
@johanstokking, wow... why do people use C? 😜 Is that also why for downlink we return size in `sendBytes()` and not the payload itself?

@Nicolasdejean I meant not to remove `port` completely but have another version (overloading?) of the method without it.

FokkeZB, 2016-09-06:
Can't we do as [this](https://arduino.land/FAQ/content/6/32/en/passing-arrays-to-functions.html#pass-multi-by-pointer)?

> These examples, while useful, are still bound to an array of a certain length and are not quite as flexible as one may need. To write a function which will accept any size array, we can start to adopt some template features. The first change we can make is to let the compiler deduce the length of the array, so we can write one piece of code for any sized array (single dimension).
> 
> ```c
> template< size_t N >
>   void Func( char (&array)[N] ){
>  
>   for( int idx = 0 ; idx < N ; ++idx ){
>     Serial.println( array[ idx ], HEX );
>   }
> }
> ```
> 
> The template parameter N will be replaced with a usable value when called with an array. And in the declaration we can replace the hard coded value with N. As this value is accessible as a constant within the function, it also eliminates the need for sizeof.

johanstokking, 2016-09-06:
No, also here you need to know the size `N` to use it. The pointer is just an address of the first element, not a reference to a struct that contains info about the array. We just have to provide the size, which is a common scenario in C.

Nicolasdejean, 2016-09-07:
If you really want to remove "int length" I think there is a way to know the size of a pointer without using sizeof.
- You end the array with a symbole like '\0'
- You count character by character until we reach the symbole like this function:

int my_strlen(const unsigned char *src)
{
  int i = 0;

  while (src[i] != '\0')
    i = i + 1;
  return (i);
}

FokkeZB, 2016-09-07:
Isn't it possible to do something like that without the last symbol? 

```c
int my_strlen(const unsigned char *src)
{
  int i = 0;
  while (src[i])
    i = i + 1;
  return (i);
}
```

johanstokking, 2016-09-07:
`0` is a valid value for everything that's not a string.

Just pass the size of the buffer. It's everywhere in C.

FokkeZB, 2016-09-07:
If that's how it's done in C, I guess we don't want to spoil folks with convenience 👅

FokkeZB, 2016-09-09:
So it seems you can calculate size from the library:
https://github.com/TheThingsNetwork/arduino-library/issues/24

Could you verify @johanstokking?

FokkeZB, 2016-09-09:
Forgive my C ignorance ;)

```
byte payload[] = { 0x00, 0x65, 0x6C, 0x6C, 0x6F };
```

length = 0 because first byte is falsy, as @johanstokking already pointed out:
https://github.com/TheThingsNetwork/arduino-library/issues/14#issuecomment-245288378

closed 2016-09-09 by FokkeZB