TheThingsNetwork/arduino-device-lib #21: set correct duty-cycle(s) after join

tftelkamp, 2016-09-08:
From the RN2483 documentation:

"The default settings consider only the three default channels (0-2), and their default duty cycle is 0.33%. If a new channel is created either by the server or by the user, all the channels (including the default ones) must be updated by the user in terms of duty cycle to comply with the ETSI regulations."

In multiple ways this is not a good implementation, but we have to work with it.

After a join, the arduino-library should set the correct duty cycles.

First, count the number of active channels (#totalchan) using the "mac get ch status" command for channels 0-15.

The duty cycle value (dcycle) to be configured on a channel is defined as: <dutyCycle> = (100/X) – 1

X is the intended duty cycle in percent. For EU we need to configure a band total duty cycle of 1%, so per channel a duty cycle of 1/#totalchan.

For each active channel, set the dcycle value to (100/#totalchan) – 1, using: 

mac set ch dcycle <channelID> <dutyCycle>

tftelkamp, 2016-09-09:
Apologies, my formula was wrong! Updated the initial comment.

Nicolasdejean, 2016-10-04:
closed by #63

closed 2016-10-04 by Nicolasdejean