TheThingsNetwork/arduino-device-lib #22: Need to configure rx2 settings and extra channels  for ABP

tftelkamp, 2016-09-08:
Set RX2 data rate to SF9:

"mac set rx2 3 869525000"

For these frequencies, configure channels 3-7:

867100000
867300000
867500000
867700000
867900000

mac set ch freq <channelID> <frequency>
mac set ch drrange <channelID> 0 5
mac set ch status <channel ID> on

The for all channels, 0-7, set:

mac set ch dcycle <channelID> 799

Additionally, change the drrange for channel 1 (868.3 Mhz) to include SF7BW250:

mac set ch drrange 1 0 6

johanstokking, 2016-09-09:
How specific is this for EU 868 frequency plan?

Proposed implementation in #31, but that looks like a lot of magic numbers to me that may be very specific to a frequency plan.

tftelkamp, 2016-09-09:
Indeed, this is specific to the TTN EU868 plan. It should only be applied on the RN2483.

Nicolasdejean, 2016-09-30:
closed by #63

Nicolasdejean, 2016-09-30:
Should we close #22 and #21 by #63 ?

Nicolasdejean, 2016-10-04:
closed by #63

closed 2016-10-04 by Nicolasdejean