TheThingsNetwork/arduino-device-lib #30: Track airtime used per day

tftelkamp, 2016-09-09:
TTN has a 30 second fair access policy. The nodes should be able to keep track of their daily airtime usage to ensure they don't violate it. 

It would be nice if we implement this in the arduino-library, as an example for people how to track this.

The node can also make decisions (e.g. spreading factor) based on remaining airtime.

cambierr, 2016-09-20:
Afaik, the uno does not have a clock nor static memory which means the only way to do this would be to track that time since the last boot, and reset in when > 1day.

Since most of the devices that will use this library are used for workshops / home dev / geek stuff, they will probably be rebooted very often... meaning that the time tracking won't work. Am I wrong ?

tftelkamp, 2016-09-20:
It's about making users aware of the airtime usage, and giving them an example of how to track it. Look at it as education. That you lose track on the Uno after a reboot is not so much a concern.

cambierr, 2016-09-21:
Ho sorry, I thought you wanted to implement a local airtime controller

closed 2016-10-18 by FokkeZB