TheThingsNetwork/arduino-device-lib #44: Hide wait process in setup

dorianamouroux, 2016-09-15:
```c

FokkeZB, 2016-09-15:
We should do this in `ttn.init()` indeed and add an argument to override the time out, defaulting to 10000.

johanstokking, 2016-09-15:
Still do it relative to the time when the method has been called, otherwise the time-out will depend on how long things took before calling init.

Also, this is only necessary if a) you're connecting a serial monitor and b) it's important that you wait with your application until your serial monitor is connected. This is fine for prototyping, but I think it's good to be able to set the library in debug mode. If debug mode is on, we wait like this on `init()`, and if not, we don't wait. Then still, if you have a serial monitor connected, you will see output, but you don't have that 10 seconds start delay.

FokkeZB, 2016-09-16:
On second thought, I'm not sure if we should do this at all. It makes it feel like a feature of the lib, while it's just an Arduino debug best practice.

johanstokking, 2016-09-16:
Exactly, so if the user wants to do it, he or she will do it in `setup()`

closed 2016-09-16 by johanstokking