TheThingsNetwork/arduino-device-lib #49: invalid_param when setting "mac set dr 5"

eliabieri, 2016-09-21:
I wondered why the module did not change to SF7 as declared as default in TheThingsnetwork.h
I built in some additional println statements and I found out that I always got the response "invalid_param"
Is this my fault or is this a bug?
Do we have to call "mac set dr 5" in another location of the code?
thank you

johanstokking, 2016-09-22:
Which RN module are you using, and which version?

This is the response of the `sys reset` call, also done in `reset()`

Nicolasdejean, 2016-09-30:
I think we solved the issue with #63 did tests with the RN2483 with the EU868 frequency plan: 
``` 
Sending: mac set ch dcycle 7 799
Sending: mac set ch drrange 1 0 6
Sending: mac set pwridx 1
Sending: mac set dr 5
Sending: mac set deveui 0004A30B001B4C9F
Sending: mac join otaa
```
There is no invalid_param. Maybe if you are using another RN module the max data rate that you can set may decrease and may cause an invalid_param.

johanstokking, 2016-09-30:
Closing this issue. @eliabieri please reopen if the problem persists. If so, please also give the RN module and version.

closed 2016-09-30 by johanstokking