TheThingsNetwork/arduino-device-lib #50: Introduce join provisioning

johanstokking, 2016-09-22:
When provisioning the device during runtime (e.g. through serial, WiFi or Bluetooth), we need to be able to set the `AppEUI` and `AppKey`, without having to join, or let it reset before join.

Proposed API:

```
bool provision(const byte appEui[8], const byte appKey[16]);
```

This method does a `mac set appeui`, `mac set deveui` and `mac save`, see [command reference](http://ww1.microchip.com/downloads/en/DeviceDoc/40001784B.pdf).

Then, you can call just `join()` with retries, as in #39

FokkeZB, 2016-09-22:
As part of this change `bool join(const byte appEui[8], const byte appKey[16])` should first call `provision(const byte appEui[8], const byte appKey[16])` and then `bool join()`

FokkeZB, 2016-09-29:
Closed by #59

closed 2016-09-29 by FokkeZB