TheThingsNetwork/arduino-device-lib #79: Design generic Measurement protobuf

johanstokking, 2016-10-07:
E.g.:

```
message Measurement {
   optional float temperature = 100;
   optional float humidity = 101;

   repeated uint32 analog_readings = 110;
}
```

FokkeZB, 2016-10-08:
I couldn't sleep tonight and had an idea for a Message class that could contain any number of values, encoded in minimal bytes based on max range and min precision.

https://github.com/TheThingsNetwork/arduino-device-lib/wiki/Proposal:-Message-Class

johanstokking, 2016-10-08:
@FokkeZB this is what protobufs will be for 👍

This allows for a generic, reusable `Message` that contains many optional fields and arrays for custom data. The device sets the fields it needs and the protobuf marshaler only puts the used values in the buffer with a tiny header. It also supports variable length integers.

I have a demo ready shortly.

johanstokking, 2016-10-08:
@FokkeZB see some example code in #81. The file https://github.com/johanstokking/ttn-protobuf-demo/blob/master/api/measurement.proto contains `required` fields, but they will be `optional` and a long list of things we can send. We can also send variable length arrays of analog readings, but putting values in the right fields helps upstream.

FokkeZB, 2016-10-08:
wow, I did quite some C coding lately, but that puzzles me 😛

FokkeZB, 2016-11-16:
Done by #108

closed 2016-11-16 by FokkeZB