TheThingsNetwork/arduino-device-lib #97: Use enum for `TTN_FP_` instead of typedef

FokkeZB, 2016-10-11:
```c
#define TTN_FP_EU868 1
#define TTN_FP_US915 2
// ..
typedef unsigned long   fp_ttn_t;
// ..
fp_ttn_t fp;
```

should become

```c
enum TTN_FP: byte
{
  TTN_FP_EU868
  TTN_FP_US915
}
// ..
TTN_FP fp;
```

so that the constructor can check for valid args:

```
TheThingsNetwork(Stream& modemStream, Stream& debugStream, TTN_FP fp, int sf = TTN_DEFAULT_SF, int fsb = TTN_DEFAULT_FSB); 
```

johanstokking, 2016-10-11:
The constructor can take `fp_ttn_t` already. Also, just name it `enum ttn_fp_t` to follow naming conventions. Uppercase is reserved for constants.

Nicolasdejean, 2016-10-11:
@FokkeZB So if they are not define anymore, do I remove TTN_FP_EU868 and TTN_FP_US915 from keywords.txt ?

FokkeZB, 2016-10-12:
Closed by #101

closed 2016-10-12 by FokkeZB