TheThingsNetwork/arduino-device-lib #104: Use strings for keys

FokkeZB, 2016-10-12:
```c
  if(lora.set_device_address("A88D5E97")) Serial.println("Address OK");
  if(lora.set_network_session_key("4230874028704F12BC3890D32D7A0355")) Serial.println("SESSIONKEY OK");
  if(lora.set_application_session_key("75B4C26938D4271F7903E3930CD14769")) Serial.println("APPKEY OK");
```

Nicolasdejean, 2016-10-24:
@FokkeZB Just to be sure, we use strings for key only for an easier comprehension for the user, right ? If so in ```lora.set_device_address``` or those other fonctions do we should convert the strings in a ``` const byte * ``` , am I understanding it correctly ?

johanstokking, 2016-10-24:
The snippet is an example from another library.

We want the user to provide the `appEui` as string instead of `byte appEui[8]`. There's no conversion needed, in fact the conversion can be omitted since the RN module wants the AppEUI, DevEUI and AppKey as hex string already. Just pass it as-is.

commit aa1e0ede (2016-10-26): Use strings for keys (closes #104)
closed 2016-10-26 by FokkeZB