TheThingsNetwork/arduino-device-lib #106: Node: Setup TTN class via Node class

FokkeZB, 2016-10-18:
Since the Node class is specific to our device we can do something like the following and hide all Serial and TTN stuff:

```c
#include <TheThingsNode.h>

const byte appEui[8] = { 0x70, 0xB3, 0xD5, 0x7E, 0xF0, 0x00, 0x00, 0x3E };
const byte appKey[16] = { 0x2A, 0xDE, 0x11, 0xDE, 0x50, 0x49, 0xB3, 0xBD, 0xFD, 0x1C, 0x75, 0x5C, 0xA4, 0xAC, 0x37, 0x7F };

TheThingsNode *node;

void setup() {
  
  node = TheThingsNode::setup();

  node->configNetwork(TTN_FP_EU868);

  node->ttn.showStatus();

  node->println("Hello");
}
```

What do you think @johanstokking ?

johanstokking, 2016-10-18:
Yes. Just use the constructor mechanism as we do with `TheThingsNetwork` as well. Also put the frequency plan there. It's a wrapper around `TheThingsNetwork` constructor and just passes the serials as we know them

amkuipers, 2016-12-06:
I saw the 1.0 code use a string, that is nice. Something that bothers me is that I have multiple nodes, one is a TTN node beta, one is a Dragino shield, one is a Sodaq One. What if there is a Register (struct array) that defines the serials for all nodes. When constructing a node, select one struct from the Register.

FokkeZB, 2016-12-06:
On second thought I think it's better to keep concerns separate and not do this. Agree @johanstokking?

johanstokking, 2016-12-08:
@amkuipers then we get a non-exhaustive list of device registrations in a communications library. I don't like a) the non-exhaustiveness as we'd always be not supporting devices and b) using more flash memory than we're already using.

johanstokking, 2016-12-08:
@FokkeZB what is the status of this issue?

FokkeZB, 2016-12-08:
I think we should close this (original) request. I'd prefer to keep this class for working with the Node's sensors and the TheThingsNetwork class for the communication. I wouldn't mix them as it would create 2 ways of sending stuff over TTN. One using TheThingsNetwork directly and one via TheThingsNode.

@amkuipers please create a new issue asking for a class with known Serials of devices. e.g. allowing the user to do:

```c
TheThingsNetwork ttn(TTN_DEVICE_SODAQONE, freqPlan);
```

closed 2016-12-08 by FokkeZB