TheThingsNetwork/arduino-device-lib #111: Bug: invalid_param in reset

FokkeZB, 2016-10-24:
I saw this in the logs:

```
Sending: mac set pwridx 1
Sending: ������������Invalid SF5
Response is not OK: invalid_param
```

Nicolasdejean, 2016-10-25:
@FokkeZB  I think its more in the ``` configureChannels ```,  What did you put in parameter in the constructor as ``` sf ```  and ``` fsb ``` ?

FokkeZB, 2016-10-25:
Nothing, and on reboot it (with the same code) worked fine again. Seems like a timing issue or some interference on the `Serial1` port? Just wanted to report in case we can catch this.

FokkeZB, 2016-10-27:
As I've understood this from @svenhaitjema this happens when commands follow each other to quickly (< ~200ms) which causes the module to see it as one.

Nicolasdejean, 2016-10-28:
So it doesn't happen very often, do you want me to put a `delay(250)` every time we do a `sendCommand()`?

FokkeZB, 2016-10-29:
That would probably hurt more than it fixes. @johanstokking ?

johanstokking, 2016-10-29:
Yeah I think so too. Let's stress test the module first (e.g. loop and set variables all the time) and see if and when it goes wrong, before adding random delays

Nicolasdejean, 2016-11-01:
I stress tested the module with a while(i < 20) loop in the ``` configureEU868() ``` (I think that is where the problem is because it is the only place where we call ``` pwridx ``` ) and a while(i < 20) loop in ``` configureChannels() ```; I also tried with a ``` while(1) ``` in ``` configureChannels()```.

But I didn't find the bug... With every loop I did every message were send correctly.

FokkeZB, 2016-11-03:
Somehow the comment got lost but @Nicolasdejean earlier said:

> I stress tested the module with a while(i < 20) loop in the  configureEU868() (I think that is where the problem is because it is the only place where we call pwridx ) and a while(i < 20) loop in [...]

So closing the issue until we run into it again

closed 2016-11-03 by FokkeZB