TheThingsNetwork/arduino-device-lib #128: First command to modem invalid after reset

FokkeZB, 2016-11-02:
Since https://github.com/TheThingsNetwork/arduino-device-lib/commit/398016e799afeb0d91e94a888e7bcc567577db40#diff-b2f2810008e6e975eb686e14fbd00f7fR21 to fix #114 the first command to the modem fails:

```
-- STATUS
EUI: invalid_param
```

This is probably because the [200ms delay](https://github.com/TheThingsNetwork/arduino-device-lib/commit/398016e799afeb0d91e94a888e7bcc567577db40#diff-b2f2810008e6e975eb686e14fbd00f7fR31) isn't long enough for the modem to see what comes next as a new command.

FokkeZB, 2016-11-02:
Even 500ms doesn't fix this. Is there another way we can fix this @svenhaitjema ?

Nicolasdejean, 2016-11-15:
@FokkeZB I don't understand how you got the problem with EUI in #114 and why init() solves it ? I can't seem to find the bug when I remove init() from `readValue()`, I think the problem here is that you write 0x55 in Serial1 with ```Serial1.write()``` with no `\r\n` at the end. So when you send the next command which here is `sys get hweui` you actually send `0x55sys get hweui` and receive an invalid_param.

FokkeZB, 2016-11-15:
Haha, I didn't either but it did work. @svenhaitjema didn't tell me to add `\r\n` but simply do a `delay()` for the modem to realize what follows is a new command. I see that https://github.com/TheThingsNetwork/arduino-device-lib/blob/239870cfb26aa57133281b335a99e3da616e23f2/src/TheThingsNetwork.cpp don't/no-longer has the `init()`. Did you remove it? 

@wienke has my Node atm so I can't test it right now. Maybe it'd be good if you get the Node until I make my way to the office again so you can test it yourself?

Nicolasdejean, 2016-11-16:
Yes I did, should I put it back and open another pull request for how to fix the init() function after the merge of https://github.com/TheThingsNetwork/arduino-device-lib/pull/130 ?
I can test it too I just need to find the node atm in the office : /.

FokkeZB, 2016-11-16:
Using #130 I no longer get this error. Probably the optimizations fixed the issue? Let's close it and reopen if we see it again, perhaps with the Node.

closed 2016-11-16 by FokkeZB