TheThingsNetwork/arduino-device-lib #134: Make `fp` argument a const in the examples

FokkeZB, 2016-11-16:
I've already seen several people run into not noticing they have to select either frequency plan:

https://github.com/TheThingsNetwork/arduino-device-lib/blob/master/examples/Send/Send.ino#L3-L10

We should change this to:

```c
const char *appEui = "0000000000000000";
const char *appKey = "00000000000000000000000000000000";
const ttn_fp_t *freqPlan = TTN_FP_EU868; // or TTN_FP_US915

#define loraSerial Serial1
#define debugSerial Serial

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);
```

@johanstokking I know you don't like to default to either plan, but it has surprised me how people didn't even notice it was a comment they needed to replace with either values. Also, with the above change the examples will not have a syntax error, which is also convenient for #121

johanstokking, 2016-11-16:
Hm. So if error A is that people don't notice it's a comment but they need to change it, and error B is that they just hit compile and upload to debug forever that it doesn't work, which error is worse? Can't we solve this with comments? RTFM?

BTW, it's not a pointer

FokkeZB, 2016-11-16:
Agree. I think it's also the place and type of (block) comment that confused people. I changed the above example to use standard `//` comment and by setting it via a const I think it's much clearer now.

johanstokking, 2016-11-16:
Yes. And the `;` to make it foolproof!

FokkeZB, 2016-11-16:
Like now?

johanstokking, 2016-11-16:
Yep

FokkeZB, 2016-11-17:
Closed by #138

closed 2016-11-17 by FokkeZB