TheThingsNetwork/arduino-device-lib #155: RFC: Make freqPlan a define?

FokkeZB, 2016-12-05:
Would you agree this change to our examples will improve the UX?

Current

```c
// Set the frequency plan
const ttn_fp_t freqPlan = // Use either TTN_FP_EU868; or TTN_FP_US915; here

#define loraSerial Serial1
#define debugSerial Serial

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);
```

Proposed

```c
#define loraSerial Serial1
#define debugSerial Serial

// Replace TTN_FP with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);
```

Nicolasdejean, 2016-12-06:
I think it's more confusing for the user, and the current one is pretty clear, isn't it ?

FokkeZB, 2016-12-06:
Why do you think it is more confusing? I think it is better for these reasons:

- It follows the same order as the arguments of the constructor
- It doesn't mix `#define` and `const`
- It doesn't require the user to know about the `ttn_fp_t` type
- The instructions to replace something rather then add are simpler

Nicolasdejean, 2016-12-06:
I agree with you about the `#define`, I just thought it was less confusing to add something (copy/paste) rather than replace it. I would have done it like this: 
`
#define freqPlan  //TTN_FP_EU868 or TTN_FP_US915
 `
And on the current one it shows an error if you didn't put anything, so it is easier for the user to notice. I don't think it shows an error on the proposed one, I may be wrong.

FokkeZB, 2016-12-06:
@Nicolasdejean it does show an error:

```
'TTN_FP' was not declared in this scope
```

I think it's better because the error is clearer then the current one:

```
expected primary-expression before 'ttn'
```

Nicolasdejean, 2016-12-06:
Ok my bad, I didn't test it right ... ok it is a better error message.

FokkeZB, 2016-12-08:
Closed by #161

closed 2016-12-08 by FokkeZB