TheThingsNetwork/arduino-device-lib #160: RFC: developer can use airtime to calculate delay/sleep

amkuipers, 2016-12-06:
In v1.0, `sendBytes` updates private airtime in class `TheThingsNetwork`. In beta, examples just send data and delay for 20s or so. In production there is a need for users to control airtime, and it would be great if the library supports it. I am not aware if TTN Uno has a RTC to 'reset daily used airtime'. But some assistance in calculating a delay/sleep would help. 
Example`: delay_ms = calc_delay( sizeof (user_data_upstream) + sizeof (user_data_downstream ))`, that uses the private `fp` and `sf`, etc, to calculate the minimum delay. 
If an RTC is available, the used airtime can be recorded per day, and the `calc_delay()` can use the remaining airtime in the calculation. 
Documentation then can have examples that use that delay, or less; multiply the delay by 2, or use something like `max( calc_delay(..), SIX_HOUR)`.

FokkeZB, 2016-12-08:
I suggest we:

- [ ] Make [`float airtime`](https://github.com/TheThingsNetwork/arduino-device-lib/blob/master/src/TheThingsNetwork.h#L45) public so that users can programmatically retrieve the total airtime to make informed decisions, e.g. to change delays. It also enables them to reset the airtime to `0`, e.g. every 24h.
- [ ] Split [`trackAirtime()`](https://github.com/TheThingsNetwork/arduino-device-lib/blob/master/src/TheThingsNetwork.cpp#L518-L527) to create separate public `calculateAirtime()` method to allow users to calculate how much a message will cost before sending it.

johanstokking, 2016-12-08:
@FokkeZB agreed

FokkeZB, 2016-12-13:
Closed by #162

closed 2016-12-13 by FokkeZB