TheThingsNetwork/arduino-device-lib #163: Create Hardware class

amkuipers, 2016-12-08:
As suggested in another issue.
Add class with known Serials of devices. e.g. allowing the user to do:

TheThingsNetwork ttn(TTN_DEVICE_SODAQONE, freqPlan);

Using commercial product (version) and/or used chip

johanstokking, 2016-12-09:
So the benefit from this is that you don't have to specify your serials.

The downside, also mentioned in [the other issue](https://github.com/TheThingsNetwork/arduino-device-lib/issues/106#issuecomment-265760170) is that we get a non-exhaustive list of devices that we need to maintain, and it all costs flash memory. Also, we need tricky compiler errors as, for example, `Serial1` (for The Things Uno) is not defined when some boards are selected.

FokkeZB, 2016-12-09:
Ah, indeed... it would fail to compile with references to non-existent `SerialX`. Closing

closed 2016-12-09 by FokkeZB