TheThingsNetwork/arduino-device-lib #179: ¿Support ESP8266?

sabas1080, 2017-01-01:
Hi

Could this work with esp8266?

Thanks

johanstokking, 2017-01-03:
The ESP8266 runs Arduino and it can probably set up a serial connection with a Microchip RN module, so I guess it does.

sabas1080, 2017-01-06:
The Version 2.1.0 does not work
I can use version 2.0.0 with some modifications work
But it seems to be the serial does not work

johanstokking, 2017-01-09:
Do you have an error message, or can you elaborate on "does not work" ?

sabas1080, 2017-01-09:
I have the next error
```
�H����hlE�DxIh��Device Information

Exception (28):
epc1=0x40202957 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000082c depc=0x00000000

ctx: cont 
sp: 3ffef1a0 end: 3ffef3d0 offset: 01a0

>>>stack>>>
3ffef340:  feefeffe 00000001 3ffe84c9 4020289c  
3ffef350:  00000000 00000001 3ffe84c9 00000005  
3ffef360:  00000005 00000000 3ffee338 402026dd  
3ffef370:  00000009 00000000 3ffee37c 40201225  
3ffef380:  3ffe84c8 feefeffe 3ffee37c 3ffee3a0  
3ffef390:  3fffdad0 00000000 3ffee338 40202726  
3ffef3a0:  3fffdad0 00000000 3ffee37c 40202a61  
3ffef3b0:  feefeffe feefeffe 3ffee398 40201740  
3ffef3c0:  feefeffe feefeffe 3ffee3b0 4020243c  
<<<stack<<<
```

with this sketch

```
#include <TheThingsNetwork.h>

#define loraSerial Serial1
#define debugSerial Serial
#define RESET 15
// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP_US915

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(9600);
  pinMode(RESET, OUTPUT);
  digitalWrite(RESET, HIGH);
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}

```
I'm trying with this sketch (Software Serial) and similar error: 

```
#include <TheThingsNetwork.h>
#include <SoftwareSerial.h>
// software serial #1: RX = digital pin 10, TX = digital pin 11
SoftwareSerial loraSerial(4, 5);

//#define loraSerial Serial1
#define debugSerial Serial
#define RESET 15
// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP_US915

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(9600);
  pinMode(RESET, OUTPUT);
  digitalWrite(RESET, HIGH);
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}
```

```
��D�L���0LDevice Information

Exception (28):
epc1=0x40202dc7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000cec depc=0x00000000

ctx: cont 
sp: 3ffef350 end: 3ffef580 offset: 01a0

>>>stack>>>
3ffef4f0:  feefeffe 00000001 3ffe84c9 40202d0c  
3ffef500:  00000000 00000001 3ffe84c9 00000005  
3ffef510:  00000005 00000000 3ffee4b8 40202ad5  
3ffef520:  00000009 00000000 3ffee530 40201225  
3ffef530:  3ffe84c8 feefeffe 3ffee530 3ffee554  
3ffef540:  3fffdad0 00000000 3ffee4b8 40202b1e  
3ffef550:  3fffdad0 00000000 3ffee530 40202f05  
3ffef560:  feefeffe feefeffe 3ffee54c 40201898  
3ffef570:  feefeffe feefeffe 3ffee560 40202590  
<<<stack<<<
```

Thanks

johanstokking, 2017-01-11:
I think this is related to #172. We're doing some dirty pointer stuff in `readLine()`. Will fix it

sabas1080, 2017-01-12:
okey, thanks I'll keep up

closed 2017-02-13 by johanstokking

sabas1080, 2017-03-14:
Hi @johanstokking 

I have tried with the latest version of the library and it seems I still have the same error or simply does not print anything

[image]

I'm trying with this sketch (Software Serial [ESPsoftwareserial](https://github.com/plerup/espsoftwareserial)) and similar error:

I 
```
#include <TheThingsNetwork.h>
#include <SoftwareSerial.h>
#include <Wire.h>

// Set your AppEUI and AppKey
const char *appEui = "AAA6789AA2311986";
const char *appKey = "000000006723FFEEBB11116734230000";

SoftwareSerial loraSerial(12, 14); // RX, TX !! labels on relay board is swapped !!

//#define loraSerial Serial1
#define debugSerial Serial

// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP_US915

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(9600);

  // Wait a maximum of 10s for Serial Monitor
  //while (!debugSerial && millis() < 10000)
  //  ;

  debugSerial.println("-- STATUS");
  ttn.showStatus();

  debugSerial.println("-- JOIN");
  ttn.join(appEui, appKey);
}

void loop()
{
  debugSerial.println("-- LOOP");

  // Prepare payload of 1 byte to indicate LED status
  byte payload[1];
  payload[0] = (digitalRead(LED_BUILTIN) == HIGH) ? 1 : 0;

  // Send it off
  ttn.sendBytes(payload, sizeof(payload));

  delay(10000);
}

```

Thanks

johanstokking, 2017-03-15:
What about the serial passthrough, see also an example of the library?

If it has to do with memory allocation of this library, then the serial passthrough should work. If it has to do with serial interfacing in general, that will crash as well.