TheThingsNetwork/arduino-device-lib #191: Add ABP initializations for Asian Frequency Plans

htdvisser, 2017-03-15:
For `AS_920_923`:

`923200000`, `923400000`, `922200000`, `922400000`, `922600000`, `922800000`, `923000000`, `922000000` DR 0-5 

For `AS_923_925`:

`923200000`, `923400000`, `923600000`, `923800000`, `924000000`, `924200000`, `924400000`, `924600000` DR 0-5

For `KR_920_923`:

`922100000`, `922300000`, `922500000`, `922700000`, `922900000`, `923100000`, `923300000` DR 0-5

johanstokking, 2017-04-11:
@jpmeijers

jpmeijers, 2017-04-17:
See https://github.com/TheThingsNetwork/arduino-device-lib/pull/199

jpmeijers, 2017-06-22:
Currently supported frequency plans are:
```
  TTN_FP_EU868,
  TTN_FP_US915,
  TTN_FP_AS920_923,
  TTN_FP_AS923_925,
  TTN_FP_KR920_923
```
This issue is therefore solved. Let's open new, separate issues for the [IN865-867](https://www.thethingsnetwork.org/wiki/LoRaWAN/Frequencies/Frequency-Plans#lorawan-frequencies_in865-867) and [AU915-928](https://www.thethingsnetwork.org/wiki/LoRaWAN/Frequencies/Frequency-Plans#lorawan-frequencies_au915-928).

closed 2017-06-22 by jpmeijers