TheThingsNetwork/arduino-device-lib #212: Onmessage payload always different

nsutter, 2017-06-15:
When i send a downlink message with mosquitto to my end-device if I send "A" ten times I will have ten times a different message in the payload. However the payload size is always correct.

mosquitto command used: 
> mosquitto_pub -t downlink/67C4D5FED21B5D13 -m '{"devaddr":"02560004","rx":1,"send_immediatly":false,"port":1,"payload":"A"}' -u <username> -P <password> -h <network>

Code used in the function called by onmessage:
> serialDebug.println("-- MESSAGE");
  serialDebug.println("Received " + String(size) + " bytes on port " + String(port) + ":");
  for (int i = 0; i < size; i++)
  {
    serialDebug.print(" ");
    serialDebug.print((payload[i]));
    lcd.print((char)payload[i]);
  }

I am doing something wrong ?
Btw i use the rn2483 chip with EU freq (868) and I tried on an arduino mega and fio.

closed 2017-06-16 by nsutter

jpmeijers, 2017-06-16:
@nsutter I see you closed this issue. Does it mean that there was a mistake on your side, or is a bug that is only rarely visible?

I'd like to update our documentation if something was unclear, or run some tests to see if everything is working as it should. Your input can help a lot.

nsutter, 2017-06-16:
Yep it's a bug on my side, I still haven't resolved it for now. It's always here but it doesn't come from the lib as I thought à the beginning.