TheThingsNetwork/arduino-device-lib #217: Send with ack retry count parameter

jpmeijers, 2017-06-22:
This is a feature request. See discussion #214.

Add a parameter to the sendBytes() function to override the default retry count of 7 when an acknowledgement is requested.
`ttn_response_t sendBytes(const uint8_t *payload, size_t length, port_t port = 1, bool confirm = false, uint8_t sf = 0, uint8_t retries = 7);`

Currently the RN2483 will retry 7 times before the blocking call to sendBytes() returns. For high spreading factors this can take a long time. For SF12, 51 byte packets: 30 minutes. Blocking for a long time, doing nothing means a big waste of energy. During this time the MCU could have been in sleep mode.

andrzejtokarski, 2017-06-25:
Having the main loop hang while waiting for the response seems a general waste of the CPU time. The situation being parallel by nature having multiple processors.
Did you guys consider another version of this method that returns after initial OK and a different one for polling the result?
I'm thinking: something similar to the Adafruit GPS library, where serial reading & parsing is done in the main loop and the state is stored by the library.

johanstokking, 2017-06-26:
Having asynchronous request/response would be very nice indeed. The problem with this serial I/O is that we still have to read from the serial somewhere, and threading is not supported on Arduino platforms. Interrupts with serial I/O neither. [SerialEvent](https://www.arduino.cc/en/Reference/SerialEvent) may have been an option, but it's not supported on Leonardo (i.e. The Things Uno) and requires it in the sketch which is not very user friendly.

Returning on OK, and reading the actual result "the next time" there is serial I/O is possible. Then you'd need to poll the library, and if you go to low power state, make sure you don't break the serial interface. We can consider this, but we have to be very careful with not going out of sync with request and response messages.