TheThingsNetwork/arduino-device-lib #219: calling reset() restore the module to factory...

alexbn71, 2017-07-07:
Hello, 

I issue an "autobaud" to comunicate with RN2483 (at 9600) but when I call the join() function it calls at the beginning the reset() function that restore RN2483 to default baud rate, thus canceling the effect of my autobaud.

If I comment out the code inside rest() that issue the reboot all works fine.

alexbn71, 2017-07-07:
I'm sorry I was working on an old version of the library but the new one has a similar problem...

alexbn71, 2017-07-07:
this it works:

`void TheThingsNetwork::reset(bool adr)
{
  // Is pretty useless this test, the baud rate could be performed every time.	
  //if (!baudDetermined)
  //{
    autoBaud();
  //}
  
  // Issue a REBOOT using the right baud rate. The response will be garbage if ne 
  // communication is not done at default baud rate and should be discarded
  size_t length = readResponse(SYS_TABLE, SYS_RESET, buffer, sizeof(buffer));
  // After REBOOT set again the module to the right baud rate.
  autoBaud();
  // And now get all infos about the module 
  length = readResponse(SYS_TABLE, SYS_TABLE, SYS_GET_VER, buffer, sizeof(buffer));

  // buffer contains "RN2xx3[xx] x.x.x ...", splitting model from version
  char *model = strtok(buffer, " ");
  debugPrintIndex(SHOW_MODEL, model);
  char *version = strtok(NULL, " ");
  debugPrintIndex(SHOW_VERSION, version);
`

johanstokking, 2017-07-17:
@alexbn71 thanks for reporting. Can you file a PR?

alexbn71, 2017-07-17:
[TheThingsNetwork_Blason.zip](https://github.com/TheThingsNetwork/arduino-device-lib/files/1152446/TheThingsNetwork_Blason.zip)

The fix is inside TheThingsNetwork::reset no other changes are necessary to get it working.

PS Ignore the two minor improvments that you could find in the code outside "TheThingsNetwork::reset" function, that are not related to this topic. 

Alex

johanstokking, 2017-07-17:
Thanks. Can you file a [pull request](https://help.github.com/articles/about-pull-requests/) here on GitHub? Otherwise I'll do it, let me know

alexbn71, 2017-07-17:
Sure, sorry...

alexbn71, 2017-07-17:
I hope I did it right, it's my first time

Alex

johanstokking, 2017-07-17:
Closed by #223

closed 2017-07-17 by johanstokking

johanstokking, 2017-07-17:
Released in https://github.com/TheThingsNetwork/arduino-device-lib/releases/tag/v2.5.4