TheThingsNetwork/arduino-device-lib #220: sleep function - overflow

savnik, 2017-07-12:
I'm working on an ATmega1284p with the RN2483A. When putting the device to sleep using the library the module does not go to sleep.
I'm using ttn.sleep(4294967295); where 4294967295 is the maximum value defined in the documentation. The result "sys sleep -1" is send to the RN2483, which results in nothing. 
If I use 2147483647 or less it works and sends "sys sleep 2147483647" and i can see the power usage go down. 
I guess somewhere there is a overflow in the library but I have not been able to spot where.

johanstokking, 2017-07-17:
So basically https://github.com/TheThingsNetwork/arduino-device-lib/blob/master/src/TheThingsNetwork.cpp#L957 results in `-1` when specifying `0xFFFFFFFF` ?

Maybe this check is wrong then? https://github.com/TheThingsNetwork/arduino-device-lib/blob/master/src/TheThingsNetwork.cpp#L948. Also I'm quite sure it has to check for `4294967295` and not `...6`.

alexbn71, 2017-07-17:
an uint32_t has a maximum value of 4,294,967,295 so the "if" has no very sense. It should be only:

`if (mseconds < 100)`

but the issue that prints "-1" is the printf...  you must use "%lu" inestead of "%ld" to print an uint32_t correctly

closed 2017-07-17 by johanstokking

johanstokking, 2017-07-17:
Released in https://github.com/TheThingsNetwork/arduino-device-lib/releases/tag/v2.5.4

PR #224 (2017-07-17): Fix sleep time overflow
Closes #220