TheThingsNetwork/arduino-device-lib #221: Wake function - feature request

savnik, 2017-07-12:
A wake function to wake the device from sleep. 
From the data sheet this function is described as:

> The module can be forced to exit from Sleep by sending a break condition followed by a
> 0x55 character at the new baud rate. Note that the break condition needs to be long
> enough not to be interpreted as a valid character at the current baud rate.

savnik, 2017-07-12:
A solution to this could be:
`  // "emulate" break condition
    Serial1.flush();

    Serial1.begin(300);
    Serial1.write((uint8_t)0x00);
    Serial1.flush();

    delay(50);

    // set baudrate
    Serial1.begin(57600);
    Serial1.write((uint8_t)0x55);
    Serial1.flush();`

That works for me with a hardware serial.

johanstokking, 2017-07-17:
In the class, we don't use serial but `Stream&`. This is so allow for interfacing that doesn't directly rely on a serial, let alone a serial implementation: Serial(N), SerialUSB, HardwareSerial, SoftwareSerial, AltSoftSerial, etc.

Also, `TheThingsNetwork` class does not declare and begin the serial at all, that's up to the caller. It would be weird to add it to a `wake()` function I think.

BTW, are you using baudrate 300 intentionally?

alexbn71, 2017-07-17:
Could be sufficient to make "void autoBaud();" a public function and call this function from code to wake up from sleep the module.

johanstokking, 2017-07-17:
@savnik @alexbn71 please advise/test keeping in mind that `TheThingsNetwork` library cannot call `begin()` as it works with the abstract `Stream`

alexbn71, 2017-07-17:
I know @johanstokking , in fact the function autoBaud() already defined in the library use Stream object.

alexbn71, 2017-07-17:
I've tested this change and it works... I moved autoBaud() into public functions section then used a call ttn.autoBaud(); to wake up the module before the expiry of the sleep time.

johanstokking, 2017-07-17:
OK, can you file a PR? Maybe it's nice to still have a method `wake()` that only calls `autoBaud()` to keep the interface clean and to reserve ourselves the freedom to do other magic stuff in there

alexbn71, 2017-07-17:
I perfectly agree with you, I do it...

johanstokking, 2017-07-18:
Closed by #225

closed 2017-07-18 by johanstokking

hallard, 2018-03-01:
it's possible to call begin casting the object as `HardwareSerial `
It's now done in PR #231 

```
  HardwareSerial *pLora = (HardwareSerial*) modemStream;
  
    // Send a 0 at lower speed to be sure always received
    // as a character a 57600 baud rate
    pLora->flush();
    pLora->begin(2400);
    pLora->write((uint8_t) 0x00);
    pLora->flush();
````

johanstokking, 2018-03-04:
This should be done in the sketch, we can't put this in the library as we can't assert `modemStream` is a `HardwareSerial`