TheThingsNetwork/arduino-device-lib #228: The Things Uno DeviceInfo serial garbage

glynhudson, 2018-01-23:
I’m running the DeviceInfo sketch on the Things Uno (with RN2433), I'm getting serial garbage as the EUI etc:

```
EUI: �
Battery: �
AppEUI: 2
DevEUI: k
Data Rate: i
RX Delay 1: )
RX Delay 2: �
```

I’ve connected the Uno via micro USB using the default sketch baudrate of 9600. I’ve also tested using a faster 115200 baud.

johanstokking, 2018-01-23:
It should indeed be 9600 baud.

Can you put the serial passthrough example on the device? See Arduino menu, Examples, TheThingsNetwork.

If you open the serial monitor, you can talk directly with the RN module. Let me know what `sys reset` says.

glynhudson, 2018-01-24:
It seems the issue is using Ubuntu to monitor the serial port of a 32u3 Leonardo. If I use the pass through sketch I can obtain the EUI. 

Thanks for your help, I received support via the forum: https://www.thethingsnetwork.org/forum/t/ttn-uno-beta-release-documentation/290/153?u=openenergymonitor

closed 2018-01-24 by glynhudson

johanstokking, 2018-01-25:
Great, thanks! Good luck