TheThingsNetwork/arduino-device-lib #236: AU915 Data Rates Incorrect

ElectronicallyE, 2018-06-27:
Hello,

I'm currently using a RN2903 in Australia with the SA1.0.3 firmware which supports the 915MHz frequency. After testing numerous factors which could be preventing me from transmitting, I discovered that my serial was giving me:

`Sending: mac set ch drrange 8 0 3`

Followed by this further down in the serial:

```
Sending: mac set dr 5
Response is not OK: invalid_param
```
This from my understanding means that the data rate range is 0-3 for the AU915 frequency and it is trying to set dr 5 which won't work.

Looking in the library and in particular thethingsnetwork.cpp, I found this:

```
void TheThingsNetwork::configureAU915(uint8_t fsb)
{
  uint8_t ch;
  uint8_t chLow = fsb > 0 ? (fsb - 1) * 8 : 0;
  uint8_t chHigh = fsb > 0 ? chLow + 7 : 71;
  uint8_t ch500 = fsb + 63;
  for (ch = 0; ch < 72; ch++)
  {
    if (ch == ch500 || (ch <= chHigh && ch >= chLow))
    {
      sendChSet(MAC_CHANNEL_STATUS, ch, "on");
      if (ch < 63)
      {
        **sendChSet(MAC_CHANNEL_DRRANGE, ch, "0 3");**
      }
    }
    else
    {
      sendChSet(MAC_CHANNEL_STATUS, ch, "off");
    }
  }
  sendMacSet(MAC_PWRIDX, TTN_PWRIDX_AU915);
}
```

From my discussion in the Slack group, AU915 supports DR0 to DR6. If the "0 3" is replaced to "0 6" it may resolve this issue. There could be other things that need to be adjusted as I am not experienced with the library.

sa-wilson, 2018-06-28:
Looking at #222 this is an issue with the RN2903 itself. I've brought it up there and we'll have to see what happens.

You should only set the DR range to [0..5] by the way, DR 6 is for the 500 kHz channel (65 in this case).

johanstokking, 2018-06-29:
Indeed, the `drrange N 0 5` is failing, which is in fact specification; this 1.0.3 module supports LoRaWAN Regional Parameters 1.0.2 rev A, and for AU915 the data rate range is 0-4 (with 4 being 500 KHz). This has changed in 1.0.2 rev B and higher.

This "1.0.3" version does not refer to the LoRaWAN (or Regional Parameters) version, but the RN firmware itself.

So the error is that DR 5 is used for sending, while it doesn't exist. I'm submitting a PR.

johanstokking, 2018-06-29:
Released in https://github.com/TheThingsNetwork/arduino-device-lib/releases/tag/v2.5.10. Arduino IDE will update automatically.

closed 2018-06-29 by johanstokking

ElectronicallyE, 2018-06-29:
Thanks Johan. Will test tonight and report back. Appreciate the rapid response from the community and yourself with this matter. 👊

leogaggl, 2018-07-02:
Can confirm that with 2.5.10 any reported invalid param errors with DR5 are now gone.

[strangely enough, the TTN Uno's have been working fine on 2.5.7 for weeks]

PR #237 (2018-06-29): Fix AU915 uplink DR
Closes #236