TheThingsNetwork/arduino-device-lib #241: ESP32 doesn't compile

jstrobel, 2018-08-09:
I am not sure if this is an error or there is a missconfiguration at my side.

I wanted to compile the Code for an Heltec-ESP32-Module and got the error:
`/Arduino/libraries/TheThingsNetwork/src/TheThingsNetwork.h:9:26: fatal error: avr/pgmspace.h: No such file or directory`

After changing in TheThingsNetwork.h from:
`#include <avr/pgmspace.h>`
to
`#include <pgmspace.h>`
it compiles.

johanstokking, 2018-08-10:
@jstrobel thanks for reporting. Did you install a library to get `pgmspace.h` on non-AVR architectures? Afaik, but I'm not an Arduino expert, the `avr/*` stuff is available with AVR architecture (which the ESP32 is not), so that file is not available indeed.

Can you try this in the header file?

```c
#ifdef ARDUINO_ARCH_AVR
#include <avr/pgmspace.h>
#else
#include <pgmspace.h>
#endif
```

And if that works, can you file a PR?

johanstokking, 2018-08-10:
I see, then it should be good, looking forward to the PR.

jstrobel, 2018-08-11:
Works for me. I did a PR.

closed 2018-08-11 by jstrobel