TheThingsNetwork/arduino-device-lib #242: Implement error handling/timeout on readLine from LoRa module

savnik, 2018-08-10:
I sometimes experience that the LoRa module(RN2483) does not send anything back after a sendBytes. The current implementation will wait forever until a return signal is send. 

`size_t TheThingsNetwork::readLine(char *buffer, size_t size)
{
  size_t read = 0;
  while (read == 0)
  {
    read = modemStream->readBytesUntil('\n', buffer, size);
  }
  buffer[read - 1] = '\0'; // set \r to \0
  return read;
}` 

I propose that a timeout is implemented to handle if there is no response from RN2483 within reasonable time.

savnik, 2018-08-10:
Yeah I see your point about returning too early. RN module as you say should work 100% of the time, however sometimes we see it locks up, not knowing why or what triggered it. I would agree that fixing the root cause would be the best solution, but unfortunately we cant access the raw firmware of the RN module to figure out what goes wrong. So then we though about how to handle this error.

If the function is returning a zero, as you suggest, it will give the application the possibility to handle the error which is an enhancement to now. 
Have you been working on such a solution already?

savnik, 2018-08-10:
I see two approaches to handle the timeout. 
1. Using milis() to measure the time spend in the while loop. 
```
size_t TheThingsNetwork::readLine(char *buffer, size_t size, uint8_t timeout = 10000) // Default timeout vallue TBD
{
  size_t read = 0;
  uint8_t start_time = milis();
  while (read == 0 && milis()-start_time<timeout)
  {
    read = modemStream->readBytesUntil('\n', buffer, size);
  }
  if(milis()-start_time<timeout)){ // If timeout is activated return 0 to inform the application
    return 0;
  }
  else{
    buffer[read - 1] = '\0'; // set \r to \0
    return read;
  }
}
```

2. Using modemStream timeout. This assumes that somewhere we set modemStream->setTimeout(x). Then if x = 1, then we will achive approximately 10s timeout as in option 1. 
```
size_t TheThingsNetwork::readLine(char *buffer, size_t size, uint8_t attempts = 10) // Default timeout attempts TBD
{
  size_t read = 0;
  while (read == 0 && attempts>0)
  {
    read = modemStream->readBytesUntil('\n', buffer, size);
    attempts--;
  }
  if(attempts<=0){ // If timeout attempts is activated return 0 to inform the application
    return 0;
  }
  else{
    buffer[read - 1] = '\0'; // set \r to \0
    return read;
  }
}
```

johanstokking, 2018-08-10:
I think we have to use a timeout, as `readBytesUntil()` should simply not return otherwise. To my understanding, the only reason for it to return `0` is on timeout.

The timeout could be set in the application sketch by the user, so that's why it's expecting and looping over `0` response now. I agree this is not nice.

Instead, I think we need to have some invalid state marker in the class, to inform the application that the RN module needs to be reset.

Can you work on a PR?

savnik, 2018-08-17:
I have created a PR. Let me know your thoughts about it.