TheThingsNetwork/arduino-device-lib #248: RN2903 and Arduino Uno not Working with Library

ElectronicallyE, 2018-12-28:
I have been trying to use the official TTN Library with my Arduino Uno wired to my Microchip RN2903 with the SA1.0.3 firmware. I am using [this](http://syncchannel.blogspot.com/2016/06/lorawan-featherwing-for-adafruit-feather.html) board for the RN2903 and have wired it to my Arduino with using a logic level converter for the RN2903 TX -> Uno RX line and two voltage dividers for the RESET line and the Uno RX -> RN2903 TX line.

I had been using the [jpmeijers](https://github.com/jpmeijers/RN2483-Arduino-Library) with success and since been testing this library with the exact same pinouts, wiring setup and AltSoftSerial. My first issue I faced was the code repeating in the `while` section of the `readLine` function which from my understanding is in the process of being resolved as documented [here](https://github.com/TheThingsNetwork/arduino-device-lib/pull/244).

My pinouts are from the UNOs perspective are:
TX (connects to RN2903 RX) = D9
RX (goes to RN2903 TX) = D8
RESET = D12

My Arduino code for all testing was based on the SendABP example, however was adjusted to support the Arduino Uno as shown below. Do note that I am using the AU915 frequency plan:

```
#include <TheThingsNetwork.h>
#include <AltSoftSerial.h>

AltSoftSerial loraSerial;

// Set your DevAddr, NwkSKey, AppSKey and the frequency plan
const char *devAddr = "ADDR GOES HERE";
const char *nwkSKey = "NWKSKEY GOES HERE";
const char *appSKey = "APPSKEY GOES HERE";

#define debugSerial Serial

// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP_AU915

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(9600);
  debugSerial.begin(57600);

  // Wait a maximum of 10s for Serial Monitor
  while (!debugSerial && millis() < 10000)
    ;

  debugSerial.println("-- PERSONALIZE");
  ttn.personalize(devAddr, nwkSKey, appSKey);
  debugSerial.println("-- STATUS");
  ttn.showStatus();
}

void loop()
{
  debugSerial.println("-- LOOP");

  // Prepare payload of 1 byte to indicate LED status
  byte payload[1];
  payload[0] = (digitalRead(LED_BUILTIN) == HIGH) ? 1 : 0;

  // Send it off
  ttn.sendBytes(payload, sizeof(payload));

  delay(10000);
}
```

I did some testing printing the read, buffer and size values by adjusting the TheThinksNetwork.cpp as follows starting on line 360:

```
size_t TheThingsNetwork::readLine(char *buffer, size_t size)
{
  debugPrintLn("Before1");
  size_t read = 0;
  debugPrintLn("Before2");
  while (read == 0)
  {
    read = modemStream->readBytesUntil('\n', buffer, size);
    debugPrint("read value: ");
    debugPrintLn(read);
    debugPrint("buffer value: ");
    debugPrintLn(buffer);
    debugPrint("size value: ");
    debugPrintLn(size);
  }
  debugPrintLn(read);
  debugPrintLn("After1");
  buffer[read - 1] = '\0'; // set \r to \0
  debugPrintLn("After2");
  return read;
}
```

My output from this is as follows:
```
-- PERSONALIZE
Before1
Before2
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
...
```
As I didn't know where to go from here, I resorted to copying the content of the three files changed in the pull request by [savnik](https://github.com/TheThingsNetwork/arduino-device-lib/pull/244/files).

These three adjusted files are:
[TheThingsNetwork.md](https://github.com/TheThingsNetwork/arduino-device-lib/blob/a17f5e494e5b1f9f8b55168aefcfcfa3f774fa6d/docs/TheThingsNetwork.md)
[TheThingsNetwork.cpp](https://github.com/TheThingsNetwork/arduino-device-lib/blob/a17f5e494e5b1f9f8b55168aefcfcfa3f774fa6d/src/TheThingsNetwork.cpp)
[TheThingsNetwork.h](https://github.com/TheThingsNetwork/arduino-device-lib/blob/a17f5e494e5b1f9f8b55168aefcfcfa3f774fa6d/src/TheThingsNetwork.h)
```

The output from these changes are below. This output took about 40 minutes, with each `mac set ch status` taking about 30 seconds each to appear.

-- PERSONALIZE
Model: Version: Sending: mac set deveui 
Sending: mac set adr off
Sending: mac set devaddr 2600169B
Sending: mac set nwkskey 973BF07958259FD82714A2A8AC2389C2
Sending: mac set appskey BA2CD38A6B7229B5AC74E3524E15F571
Sending: mac set ch status 0 off
Sending: mac set ch status 1 off
Sending: mac set ch status 2 off
Sending: mac set ch status 3 off
Sending: mac set ch status 4 off
Sending: mac set ch status 5 off
Sending: mac set ch status 6 off
Sending: mac set ch status 7 off
Sending: mac set ch status 8 on
Sending: mac set ch drrange 8 0 3
Sending: mac set ch status 9 on
Sending: mac set ch drrange 9 0 3
Sending: mac set ch status 10 on
Sending: mac set ch drrange 10 0 3
Sending: mac set ch status 11 on
Sending: mac set ch drrange 11 0 3
Sending: mac set ch status 12 on
Sending: mac set ch drrange 12 0 3
Sending: mac set ch status 13 on
Sending: mac set ch drrange 13 0 3
Sending: mac set ch status 14 on
Sending: mac set ch drrange 14 0 3
Sending: mac set ch status 15 on
Sending: mac set ch drrange 15 0 3
Sending: mac set ch status 16 off
Sending: mac set ch status 17 off
Sending: mac set ch status 18 off
Sending: mac set ch status 19 off
Sending: mac set ch status 20 off
Sending: mac set ch status 21 off
Sending: mac set ch status 22 off
Sending: mac set ch status 23 off
Sending: mac set ch status 24 off
Sending: mac set ch status 25 off
Sending: mac set ch status 26 off
Sending: mac set ch status 27 off
Sending: mac set ch status 28 off
Sending: mac set ch status 29 off
Sending: mac set ch status 30 off
Sending: mac set ch status 31 off
Sending: mac set ch status 32 off
Sending: mac set ch status 33 off
Sending: mac set ch status 34 off
Sending: mac set ch status 35 off
Sending: mac set ch status 36 off
Sending: mac set ch status 37 off
Sending: mac set ch status 38 off
Sending: mac set ch status 39 off
Sending: mac set ch status 40 off
Sending: mac set ch status 41 off
Sending: mac set ch status 42 off
Sending: mac set ch status 43 off
Sending: mac set ch status 44 off
Sending: mac set ch status 45 off
Sending: mac set ch status 46 off
Sending: mac set ch status 47 off
Sending: mac set ch status 48 off
Sending: mac set ch status 49 off
Sending: mac set ch status 50 off
Sending: mac set ch status 51 off
Sending: mac set ch status 52 off
Sending: mac set ch status 53 off
Sending: mac set ch status 54 off
Sending: mac set ch status 55 off
Sending: mac set ch status 56 off
Sending: mac set ch status 57 off
Sending: mac set ch status 58 off
Sending: mac set ch status 59 off
Sending: mac set ch status 60 off
Sending: mac set ch status 61 off
Sending: mac set ch status 62 off
Sending: mac set ch status 63 off
Sending: mac set ch status 64 off
Sending: mac set ch status 65 on
Sending: mac set ch status 66 off
Sending: mac set ch status 67 off
Sending: mac set ch status 68 off
Sending: mac set ch status 69 off
Sending: mac set ch status 70 off
Sending: mac set ch status 71 off
Sending: mac set pwridx 5
Sending: mac set retx 7
Sending: mac set dr 3
Sending: mac join abp 
Personalize accepted. Status: 
-- STATUS
EUI: 
Battery: 
AppEUI: 
DevEUI: 
Data Rate: 
RX Delay 1: 
RX Delay 2: 
-- LOOP
Sending: mac tx uncnf 1 01
Successful transmission
-- LOOP
Sending: mac tx uncnf 1 01
Successful transmission
-- LOOP
Sending: mac tx uncnf 1 01
Successful transmission
-- LOOP
Sending: mac tx uncnf 1 01
Successful transmission
-- LOOP
Sending: mac tx uncnf 1 01
Successful transmission
```

The node began transmitting and did it rather consistently, which appeared in the gateway console, however nothing appeared within the node console. I checked to make sure that my credentials were correct and they were, so on the TTN side everything should have worked.

Also note the blank value under `STATUS` which I can only assume should contain information.

The above channels also look [correct](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html) being from channels 8-15 and then 65, so there is nothing wrong with the frequency plan either.

So the current problems are as follows:
1. Looping in the `readLine` function
2. Long initialisation setup time for adjusted library
3. Missing values under `STATUS`

ElectronicallyE, 2018-12-29:
Hi @johanstokking,

Using @jpmeijers library I am able to execute commands on the RN2903.

In addition, I have used the following code without either @jpmeijers or the TTN library and it succeeded in send a message to the RN2903 and sending one back. On my board a LED is connect to GPIO10, so for testing I flicked it on and off.

Below is my code. It's not pretty, but it shows that it works with my current setup:
```
#include <AltSoftSerial.h>

AltSoftSerial loraSerial;

void setup()
{
  Serial.begin(57600); //serial port to computer
  loraSerial.begin(9600); //serial port to radio
  Serial.println("Startup");
  initialize_radio();
  delay(2000);
  Serial.println("RN2xx3 firmware version:");
  Serial.println(sysver());
}

void loop()
{
  sendRawCommand(F("sys set pindig GPIO10 1"));
  delay(1000);
  sendRawCommand(F("sys set pindig GPIO10 0"));
  delay(1000);
}

void initialize_radio()
{
  //reset rn2483
  pinMode(12, OUTPUT);
  digitalWrite(12, LOW);
  delay(500);
  digitalWrite(12, HIGH);

  delay(100); //Wait for the RN2xx3's startup message
  loraSerial.flush();
  
  autobaud(); //Autobaud the rn2483 module to 9600. The default would otherwise be 57600.
}

String sendRawCommand(String c)
{
  while(loraSerial.available())
  {
    loraSerial.read();
  }
  loraSerial.println(c);
  String ret = loraSerial.readStringUntil('\n');
  ret.trim();
  return ret;
}

String sysver()
{
  String ver = sendRawCommand(F("sys get ver"));
  ver.trim();
  return ver;
}

void autobaud()
{
  String response = "";
  while (response=="")
  {
    delay(1000);
    loraSerial.write((byte)0x00);
    loraSerial.write(0x55);
    loraSerial.println();
    loraSerial.println("sys get ver");
    response = loraSerial.readStringUntil('\n');
  }
}
```

johanstokking, 2018-12-31:
So the issue is that this bare code is faster than when you use the library?

That's pretty weird...

ElectronicallyE, 2019-01-01:
I'd say so, but the bigger issue is that the library simply does not play nice with the official TTN library.

johanstokking, 2019-01-02:
Which library does not play nice with the TTN library?

The only obvious thing I see in your code is that you swapped the baud rates for debug and the radio; if you use 9600 for the RN module instead of 57600 then everything is clearly slower.

ElectronicallyE, 2019-01-02:
Sorry, I meant to say board. ;) The baud rate I normally use is 9600, I'll try changing that.

johanstokking, 2019-01-21:
Any news @ElectronicallyE ?

ElectronicallyE, 2019-01-21:
Hey @johanstokking,

Changed the baud rate 9600 and no difference. I did the serial printing of what is happening in the newly updated library using the following code:

size_t TheThingsNetwork::readLine(char *buffer, size_t size, uint8_t attempts)
{
  Serial.println("BEFORE");
  size_t read = 0;
  while (!read && attempts--)
  {
    read = modemStream->readBytesUntil('\n', buffer, size);
    debugPrint("read value: ");
    debugPrintLn(read);
    debugPrint("buffer value: ");
    debugPrintLn(buffer);
    debugPrint("size value: ");
    debugPrintLn(size);
  }
  if (attempts<=0)
  { // If attempts is activated return 0 and set RN state marker
    this->needsHardReset = true; // Inform the application about the radio module is not responsive.
    debugPrintLn("No response from RN module.");
    return 0;
  }
  debugPrintLn(read);
  debugPrintLn("After1");
  buffer[read - 1] = '\0'; // set \r to \0
  debugPrintLn("After2");
  buffer[read - 1] = '\0'; // set \r to \0
  return read;
}

 And got this output:

-- PERSONALIZE
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Model: Version: read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Sending: mac set deveui 
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Sending: mac set adr off
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Sending: mac set devaddr 2600169B
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Sending: mac set nwkskey 973BF07958259FD82714A2A8AC2389C2
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
read value: 0
buffer value: 
size value: 512
0
After1
After2
Sending: mac set appskey BA2CD38A6B7229B5AC74E3524E15F571
read value: 0
buffer value: 
size value: 512

It keeps going on, so I'm assuming it'll do what was happening before when I had previously change the files.

johanstokking, 2019-01-22:
Which change did you actually apply? Because I still see baud rate `9600` for the LoRa serial, while that should be `57600`.

ElectronicallyE, 2019-01-22:
I changed both to 9600. I’ve updated it in my post, because that’s what I actually did.

johanstokking, 2019-01-22:
OK. It is normal that you experience slower communication on a lower baud rate. You can and should use 57600 for the RN module.

ElectronicallyE, 2019-01-23:
I believe the problem is still present. It seems to me that the print out takes a lot longer than necessary at about 40 minutes, and yet, I still do not see a transmission on the TTN Console.

I have used other code with the [JP Meijers Library](https://github.com/jpmeijers/RN2483-Arduino-Library ) where I communicate at 57600, so I don't think that is the problem.

johanstokking, 2019-01-23:
@jpmeijers any idea?

johanstokking, 2019-07-08:
@ElectronicallyE did you figure out in the meantime why the TTN library doesn't work?

ElectronicallyE, 2019-07-25:
@johanstokking, unfortunately I haven't. Completely slipped out of my mind. I won't be able to get back into it till the end of the year, but when I make progress, I'll be sure to get back in touch.

johanstokking, 2019-08-06:
OK please reopen when the issue is still there.

closed 2019-08-06 by johanstokking

brolly759, 2020-09-05:
Was this ever fixed? I tried the RN2903 module and still cant get this to work.

ElectronicallyE, 2020-09-05:
Hey @brolly759 I didn’t end up resolving this issue as I no longer had the need to develop a board I was designing which used the RN2903. Have a look at this library, which I believe I got working, but was stuck on SF12 packet size limits: https://github.com/jpmeijers/RN2483-Arduino-Library

brolly759, 2020-09-06:
I tested that library out as well.  I couldn't do more than 11 byte payloads. Also running into issues trying to get network status on there. In their code they have a switch statement for the RN2903, I guess I could compare the TTN library verse theirs to see the differences. Maybe I will just go with a different UART Lora module. ;)

ElectronicallyE, 2020-09-06:
Yep, that’s about where I stopped. Was just putting a heap of time into it with very little progress and by the end it was more effective to just go with a MKR WAN 1310.

Oddly enough, the TTN Uno as far as I know didn’t have these problems, which makes me suspect it was something about the Uno and the RN2903.

brolly759, 2020-09-06:
MKR WAN 1310. That uses a Murata module and looks like SPI to control it over an SAMD21 processor. Not as easy as UART.

brolly759, 2020-09-06:
Nevermind, looking at the source code it looks like UART. Still a bit overkill if I wanted to use an ATMEGA328P.

brolly759, 2020-09-06:
haha, I got it to work! That was easier than I thought. Make sure the buadrate for both serial ports is set to 9600. 57600 is too fast and data can get corrupted with the softwareserial library. Then all you had to do was call ttn.reset(); after loraSerial.begin(9600); debugSerial.begin(9600); 

My theory is this, the autobuad() function isnt called after the serial ports are initialized and we have a buadrate mismatch.
The OTAA example works fine.

jpmeijers, 2020-09-06:
@brolly759 your solution, how much does it differ from that is documented at https://www.thethingsnetwork.org/docs/devices/arduino/usage.html#serial-ports?

If the documentation is wrong or incomplete, please let us know, or file a pull request to fix it.

brolly759, 2020-09-06:
@jpmeijers First off, great work! 
The documentation isn't wrong at all. 
_The only thing you need to add, at least for the "Arduino Uno, Arduino Nano or other devices using SoftwareSerial" section is ttn.reset(); in the setup() function after the serial.begin statements._ 

Reasoning: because we are changing the baudrate from 57600 to 9600 we need to run the autobaud function after the serial ports are initialized. The autobaud() function is private in the library and the only time its called is in ttn.reset(); Its also good habit to reset the Microchip module on startup anyways.