TheThingsNetwork/arduino-device-lib #249: SODAQ LoRaExplorer not compiling due to missing pgmspace.h

proffalken, 2019-01-16:
I'm trying to compile the basic SODAQ example from the LoRaWAN Academy, but I'm hitting an issue when trying to compile.

The code is simple:

```
#include <TheThingsNetwork.h>

#define loraSerial Serial2
#define debugSerial SerialUSB

// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915
#define freqPlan TTN_FP_EU868

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(9600);
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}
```

However the error I'm seeing is:

```Arduino: 1.8.7 (Linux), Board: "SODAQ ExpLoRer"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/mmw/.arduino15/packages -hardware /home/mmw/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/mmw/.arduino15/packages -libraries /home/mmw/Arduino/libraries -fqbn=SODAQ:samd:sodaq_explorer -vid-pid=0X2341_0X804D -ide-version=10807 -build-path /tmp/arduino_build_88520 -warnings=none -build-cache /tmp/arduino_cache_27221 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arduinoOTA.path=/home/mmw/.arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=/home/mmw/.arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.avrdude.path=/home/mmw/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.avrdude-6.3.0-arduino8.path=/home/mmw/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.openocd.path=/home/mmw/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=/home/mmw/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.bossac.path=/home/mmw/.arduino15/packages/arduino/tools/bossac/1.7.0 -prefs=runtime.tools.bossac-1.7.0.path=/home/mmw/.arduino15/packages/arduino/tools/bossac/1.7.0 -verbose /home/mmw/Arduino/LoRaAcademy/SodaqDeviceInfo/SodaqDeviceInfo.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/mmw/.arduino15/packages -hardware /home/mmw/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/mmw/.arduino15/packages -libraries /home/mmw/Arduino/libraries -fqbn=SODAQ:samd:sodaq_explorer -vid-pid=0X2341_0X804D -ide-version=10807 -build-path /tmp/arduino_build_88520 -warnings=none -build-cache /tmp/arduino_cache_27221 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arduinoOTA.path=/home/mmw/.arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=/home/mmw/.arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.avrdude.path=/home/mmw/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.avrdude-6.3.0-arduino8.path=/home/mmw/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.openocd.path=/home/mmw/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=/home/mmw/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.bossac.path=/home/mmw/.arduino15/packages/arduino/tools/bossac/1.7.0 -prefs=runtime.tools.bossac-1.7.0.path=/home/mmw/.arduino15/packages/arduino/tools/bossac/1.7.0 -verbose /home/mmw/Arduino/LoRaAcademy/SodaqDeviceInfo/SodaqDeviceInfo.ino
Using board 'sodaq_explorer' from platform in folder: /home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19
Using core 'arduino' from platform in folder: /home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19
Detecting libraries used...
/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SODAQ_EXPLORER -DARDUINO_ARCH_SAMD -D__SAMD21J18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"SODAQ\"" "-DUSB_PRODUCT=\"SODAQ ExpLoRer\"" -I/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/ -I/home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19/cores/arduino -I/home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19/variants/sodaq_explorer /tmp/arduino_build_88520/sketch/SodaqDeviceInfo.ino.cpp -o /dev/null
/home/mmw/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SODAQ_EXPLORER -DARDUINO_ARCH_SAMD -D__SAMD21J18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"SODAQ\"" "-DUSB_PRODUCT=\"SODAQ ExpLoRer\"" -I/home/mmw/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/mmw/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/ -I/home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19/cores/arduino -I/home/mmw/.arduino15/packages/SODAQ/hardware/samd/1.6.19/variants/sodaq_explorer -I/home/mmw/Arduino/libraries/TheThingsNetwork/src /tmp/arduino_build_88520/sketch/SodaqDeviceInfo.ino.cpp -o /dev/null
In file included from /home/mmw/Arduino/LoRaAcademy/SodaqDeviceInfo/SodaqDeviceInfo.ino:1:0:
/home/mmw/Arduino/libraries/TheThingsNetwork/src/TheThingsNetwork.h:13:22: fatal error: pgmspace.h: No such file or directory
 #include <pgmspace.h>
                      ^
compilation terminated.
Using library TheThingsNetwork at version 2.5.12 in folder: /home/mmw/Arduino/libraries/TheThingsNetwork 
exit status 1
Error compiling for board SODAQ ExpLoRer.```

I thought it might be related to #241 , however the patch appears to be present in the header file, so I'm not sure what's causing this!

johanstokking, 2019-01-16:
Hmm, I'm afraid we introduced it in #241.

Can you confvirm it works with `avr/pgmspace.h`?

proffalken, 2019-01-18:
Yup, hard-coding it to that value works fine.

johanstokking, 2019-01-19:
OK that's regression from #243.

Can you file a PR to fix this? We need a compiler directive again, but this time taking into account this platform.

proffalken, 2019-01-21:
Sure, I'll get to it ASAP.

proffalken, 2019-01-22:
OK, I'm now thoroughly confused - I went to fork/patch/PR and found out that the logic was already there.  I then check the Arduino libraries manager and saw that there was an update, and on updating it pulled in the code and compiled successfully.

It looks like the relevant commit is https://github.com/TheThingsNetwork/arduino-device-lib/commit/b967fc4a773873bed344ac89994de31242a744c3 but I've no idea why my libraries weren't showing as out of date in the IDE until now.

Anyway, it's working with the latest libraries in the IDE, so I'll "Close - Won't Fix" this now, and try not to look too embarrassed at the conference next week! :joy:

closed 2019-01-22 by proffalken