TheThingsNetwork/arduino-device-lib #255: showStatus() with other Arduino boards

Erik-Koning, 2019-06-07:
Hello,

Im using this [lora node board](https://www.thethingsnetwork.org/forum/t/solved-elecrow-lora-node-radio-v1-0/24201/3) but i've been unsuccessful with finding its EUI using the showStatus() method. It obviously must have a different pin layout that the things network nodes. Where would i change the pin layout in this library's code to enable it?

This is the output of the deviceInfo.ino:
`
---------Device Information

EUI: 
Battery: 
AppEUI: 
DevEUI: 
Data Rate: 
RX Delay 1: 
RX Delay 2: 

Use the EUI to register the device for OTA
`

Erik-Koning, 2019-06-10:
The [LMIC_arduino library ](https://github.com/matthijskooijman/arduino-lmic) has a pin mapping const variable as follows:
`// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 6,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};`

Is there something similar in this repo that I should set to generate device information?

johanstokking, 2019-06-11:
This library only works with Microchip RN2xx4 modules.

closed 2019-06-11 by johanstokking