TheThingsNetwork/arduino-device-lib #262: Working with Arduino Pro Mini + RN2483

LasaleFamine, 2020-06-05:
I'm going to build some small devices with this setup: https://github.com/rac2030/MakeZurich/wiki/Hello-Lora-with-Arduino-Pro-mini-and-Microchip-RN2483

I'm wondering if I can use the lib with the Arduino Pro Mini, I assume the `Serial1` won't be available. Do you have some hints about this?

jpmeijers, 2020-06-05:
The library should work. You can try and use software serial at 9600 baud.

LasaleFamine, 2020-06-05:
Thanks for your reply. I'm gonna try asap. Do you think it's worth to add an example in to the example folder regarding this specific case? Or I'm missing an example already there anyway?

LasaleFamine, 2020-06-05:
I've tried with an Arduino Uno (that should be pretty the same) but I can't make it work. Note that the example on the link works with the current configuration and I can receive a response from the module. But trying the software serial for `DeviceInfo` example I always receive "No response from RN module". Here is the modified example:

```c
#include <TheThingsNetwork.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(7, 8); // RX, TX
#define loraSerial mySerial
#define debugSerial Serial

#define freqPlan TTN_FP_EU868

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(9600);
  debugSerial.begin(115200);
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}
```

I have also tried manually resetting the module as per the example on the link, but I get the same result. Do you know what I'm doing wrong?

LasaleFamine, 2020-06-06:
Sorry for the triple post, but at the end I've got it working. This is the working sketch:

```c
#include <TheThingsNetwork.h>
#include <SoftwareSerial.h>

/*
  RN2483 pin name <--> Arduino pin number
  UART_TX (6)   <-->   10
  UART_RX (7)   <-->   11
  RESET (32)    <-->   12
  VDD (34)      <-->   3.3V
  GND (33)      <-->   Gnd
*/

#define RST  12
SoftwareSerial loraSerial(10, 11); // RX, TX
#define debugSerial Serial

#define freqPlan TTN_FP_EU868

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(9600);
  debugSerial.begin(9600);
  
  // Reset rn2483
  pinMode(RST, OUTPUT);
  digitalWrite(RST, HIGH);
  digitalWrite(RST, LOW);
  delay(500);
  digitalWrite(RST, HIGH);
  
  ttn.wake();
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}
```

A side note: I've tried also with `ttn.reset()` and `ttn.resetHard(RST)` but both don't have the same effect as the reset method used above. I'm thinking if this is a bug of the `reset` or `resetHard` implementation. I'm leaving this opened for further investigation. Also let me know if I can help someway! Thank you.

jpmeijers, 2020-06-07:
From TTN's side I believe it is documented well enough:
https://www.thethingsnetwork.org/docs/devices/arduino/usage.html#serial-ports

The docs, plus your examples in this issue is good enough and should guide any future users in the right direction.

jpmeijers, 2020-06-07:
Regarding the reset, maybe open a dedicated issue for that.

closed 2020-06-07 by jpmeijers