TheThingsNetwork/arduino-device-lib #265: ESP32 Crashes With TheThingsNetwork.h

kevin192291, 2020-07-10:
The Device Info sketch configured like so seems to be causing the ESP32 to go into infinite loop.
Has anyone attempted to get this working? Any word on why this happens?
I am using the ESP32's hardware serial port #2. I have attempted to get this working with the
library RN2xx3.h, it will work to a point.
What pin is TheThingsNetwork.h pointed at? or is it even configured? I checked the code and can't
seem to find where this is configured, there is a hardReset function, but that takes a pin as input.

```
#include <TheThingsNetwork.h>

#define loraSerial Serial2
#define debugSerial Serial
#define freqPlan TTN_FP_US915

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(115200);
}

void loop()
{
  debugSerial.println("Device Information");
  debugSerial.println();
  ttn.showStatus();
  debugSerial.println();
  debugSerial.println("Use the EUI to register the device for OTAA");
  debugSerial.println("-------------------------------------------");
  debugSerial.println();

  delay(10000);
}
```

jpmeijers, 2020-07-13:
The pins that are used are defined by the platform's Serial definitions. In other words if you are using Serial2, the pins that are defined for Serial2 by the ESP32 Arduino board definitions are used.

kevin192291, 2020-07-13:
Hello, yes I know this. I have the module wired correctly. 
I have attempted to use a different library RN2xx3.h and can successfully transmit small amounts of data.
TheThingsNetwork.h also works great with an Arduino Due, I just can't seem to get this to work with the ESP32, I am
wondering if TheThingsNetwork.h library is using an incompatible library somewhere? Has anyone got this working at
any point? Also, I have attempted to use SoftwareSerial instead of the hardware Serial2 port, and it fails in the same way.

Finally, I am not using all of the serial ports at the same time (due to that bug)

closed 2020-08-18 by kevin192291