TheThingsNetwork/arduino-device-lib #267: RN2903 using freqPlan TTN_FP_AS920_923 fails

kevin192291, 2020-08-18:
I am attempting to use The Things Uno with the Asian 920-923 range, however
I am getting the errors:
```
Sending: mac save 
Sending: mac set adr off
Sending: mac set rx2 2 923200000
Response is not OK: invalid_param
Sending: mac set ch dcycle 0 799
Response is not OK: invalid_param
Sending: mac set ch dcycle 1 799
Response is not OK: invalid_param
Sending: mac set ch dcycle 2 799
Response is not OK: invalid_param
Sending: mac set ch freq 2 922000000
Response is not OK: invalid_param
Sending: mac set ch drrange 2 0 5
Response is not OK: invalid_param
Sending: mac set ch status 2 on
Sending: mac set ch dcycle 3 799
Response is not OK: invalid_param
Sending: mac set ch freq 3 922200000
Response is not OK: invalid_param
Sending: mac set ch drrange 3 0 5
Response is not OK: invalid_param
Sending: mac set ch status 3 on
Sending: mac set ch dcycle 4 799
Response is not OK: invalid_param
Sending: mac set ch freq 4 922400000
Response is not OK: invalid_param
Sending: mac set ch drrange 4 0 5
Response is not OK: invalid_param
Sending: mac set ch status 4 on
Sending: mac set ch dcycle 5 799
Response is not OK: invalid_param
Sending: mac set ch freq 5 922600000
Response is not OK: invalid_param
Sending: mac set ch drrange 5 0 5
Response is not OK: invalid_param
Sending: mac set ch status 5 on
Sending: mac set ch dcycle 6 799
Response is not OK: invalid_param
Sending: mac set ch freq 6 922800000
Response is not OK: invalid_param
Sending: mac set ch drrange 6 0 5
Response is not OK: invalid_param
Sending: mac set ch status 6 on
Sending: mac set ch dcycle 7 799
Response is not OK: invalid_param
Sending: mac set ch freq 7 923000000
Response is not OK: invalid_param
Sending: mac set ch drrange 7 0 5
Response is not OK: invalid_param
Sending: mac set ch status 7 on
Sending: mac set pwridx 1
Response is not OK: invalid_param
Sending: mac set retx 7
Sending: mac set dr 5
Response is not OK: invalid_param
```

on startup,

Here is my code that I simply ran from the example. I have omitted my keys:
```
#include <TheThingsNetwork.h>
#include <TheThingsMessage.h>

// Set your AppEUI and AppKey
const char *appEui = "secret";
const char *appKey = "very secret";

#define loraSerial Serial1
#define debugSerial Serial

// Replace REPLACE_ME with TTN_FP_EU868 or TTN_FP_US915 or TTN_FP_AS920_923
#define freqPlan TTN_FP_AS920_923

TheThingsNetwork ttn(loraSerial, debugSerial, freqPlan);

devicedata_t data = api_DeviceData_init_default;

void setup()
{
  loraSerial.begin(57600);
  debugSerial.begin(115200);

  // Wait a maximum of 10s for Serial Monitor
  while (!debugSerial && millis() < 10000)
    ;

  debugSerial.println("-- STATUS");
  ttn.showStatus();

  debugSerial.println("-- JOIN");
  ttn.join(appEui, appKey);

  // Select what fields to include in the encoded message
  data.has_motion = true;
  data.has_water = false;
  data.has_temperature_celcius = true;
  data.has_temperature_fahrenheit = true;
  data.has_humidity = true;
}

void loop()
{
  debugSerial.println("-- LOOP");

  // Read the sensors
  data.motion = true;

  // Encode the selected fields of the struct as bytes
  byte *buffer;
  size_t size;
  TheThingsMessage::encodeDeviceData(&data, &buffer, &size);

  ttn.sendBytes(buffer, size);

  delay(10000);
}
```

johanstokking, 2020-11-12:
Can you try with the passthrough example, and issuing commands directly as specified in the data sheet?

RCPilot1604, 2022-03-04:
@kevin192291 I'm having very similar issues as you. I wrote my sketch and debug serial output [on the TTN forum here](https://www.thethingsnetwork.org/forum/t/difference-between-versions-of-rn2903a-i-rmxx/54533/4). 

I suspect the supported frequency plans are hard-coded into the chip because changing the freqPlan TTN_FP_AS920_923
to TTN_FP_US915 seemed to solve all the invalid_param errors... 

I bought my chip [here](https://sg.element14.com/microchip/rn2903a-i-rm105/transceiver-module-300kbps-928mhz/dp/2932052?st=rn2903) and am not really that surprised since it did say "North-America" in the description... 

Will be trying [this one](https://sg.element14.com/microchip/rn2903a-i-rm103/rf-transceiver-module-300kbps/dp/2851711?st=rn2903) out and will let you know.

kevin192291, 2022-03-04:
@RCPilot1604 It has been quite some time since I was trying to use this chip, I have found there are alternatives. After talking to Microchip directly, they told me that they have never released an official fw for as923. And, to expand on this, this chip lacks a Telec certification, so if you want it to work for hobby purposes, you will be okay, but it is a no-go for building actual products.
I have been using RAK chips lately, these are just as, if not more easy to work with. [RAK3172](https://store.rakwireless.com/products/wisduo-breakout-board-rak3272s?variant=40104492957894) This is certified, it supports LoRaWAN Class A, B, and C, and is a serial connection just like the RN2903.

closed 2022-03-04 by kevin192291