TheThingsNetwork/arduino-device-lib #269: Error when compiling DeviceInfo example for ESP8266 board

vonsivers, 2020-09-19:
Hi there,
when trying to compile the DeviceInfo example for Board "Generic ESP8266 module" I get the following error.

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), no dtr (aka ck), 26 MHz, 40MHz, DOUT (compatible), 4MB (FS:3MB OTA:~512KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\Moritz\Documents\Arduino\libraries\TheThingsNetwork\src\TheThingsNetwork.cpp: In member function 'ttn_response_t TheThingsNetwork::poll(port_t, bool)':

C:\Users\Moritz\Documents\Arduino\libraries\TheThingsNetwork\src\TheThingsNetwork.cpp:652:41: error: 'class Stream' has no member named 'getTimeout'

       long timeout = this->modemStream->getTimeout();

jpmeijers, 2020-09-20:
So it seems like the ESP8266 Arduino libraries are not complete and do not implement the getTimeout() function. A proper fix would be to add this function to the ESP libraries. A workaround could be to "guess" the value and just assume we are close enough. Normally a timeout would be in the order of 1 second.

vonsivers, 2020-09-23:
So I changed the following line in TheThingsNetwork.cpp 

```
//long timeout = this->modemStream->getTimeout();
long timeout = 1000; in your library to
```

I can compile the code now, however, after upload the MCU crashes with the following exception

i⸮JW⸮ai⸮J⸮⸮CH,E⸮Device Information

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40206f26 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00008014 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  80008015 feefeffe feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe 00000009  
3ffffed0:  00000001 00000000 3ffee3e8 402012e4  
3ffffee0:  00000760 feefeffe feefeffe feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe 00000100  
3fffff00:  00000001 3ffef38c 00000020 40100900  
3fffff10:  40202ba1 feefeffe feefeffe feefeffe  
3fffff20:  3ffe8699 00000000 6553000a 40202b09  
3fffff30:  3ffe8699 3ffee3f7 3ffee3e8 402014f1  
3fffff40:  3ffe8699 40203079 3ffee3e8 40203910  
3fffff50:  00000200 3ffee3f7 3ffee3e8 40201356  
3fffff60:  00000005 00000005 3ffe8697 402014d0  
3fffff70:  402014c4 3ffee620 3ffe8697 3ffee688  
3fffff80:  3fffdad0 3ffee3f7 3ffee3e8 402013b3  
3fffff90:  3fffdad0 00000000 3ffee620 40201085  
3fffffa0:  feefeffe feefeffe 3ffee648 40201dac  
3fffffb0:  feefeffe feefeffe 3ffe84e0 40100b85  
<<<stack<<<

jpmeijers, 2020-09-24:
At this point maybe try a tool like https://github.com/me-no-dev/EspExceptionDecoder/tree/1.1.0 to decode the above crash into a more usable format. Hopefully that will indicate on which line the program is crashing.