TheThingsNetwork/arduino-device-lib #274: TTN strcpy_P and pgmstrcmp start to fail after including more code elsewhere in the application

bsc-itcl, 2021-02-24:
Hello,

I am using this library on a custom board based on Arduino Zero (same SAMD microcontroller) with the RN2483 LoRa transceiver. TTN was working perfectly until, at some point after adding extra code in other files, LoRa initialization started to fail. The symptoms were that the buffers were corrupt since the beginning (reset function), checked with the debug Serial and the oscilloscope.

At the beginning I could work around it commenting out some parts of my other code (floats, Serial.prints, etc) but in the end this was the bottleneck.

Finally after a lot of debugging with the oscilloscope I found out the problem was in the strcpy_P function: I replaced all strcpy_P for strcpy and e.g. pgmstrcmp(buffer, CMP_ACCEPTED) for strcmp(buffer, compare_table[CMP_ACCEPTED]).
I guess this is not optimal, but for now it's working. I can try to give more information if anyone cares, I just post this to let everyone know my solution and because I'm curious about the root cause and eager to learn.

Regards

jpmeijers, 2021-02-24:
It sounds like you are running out of memory (RAM) during runtime, corrupting your buffers.

Because this library fits and works correctly on AVR (Arduino Leonardo and Uno), and the SAMD your are using has much more memory that the AVR, I doubt that the issue is in this library. Most likely you are defining a buffer in your code that is very big (512 bytes or more), which eats up all the memory, corrupting this library's buffers.

There could be other causes too, but this is the most likely one.

bsc-itcl, 2021-02-25:
Hi @jpmeijers thanks for your reply.

You are right, it looks like an out-of-RAM problem. I don't have any big buffer or array, however I am using several other libraries for the different sensors which may sum up more variables.

However, the solution I got (and that is working well so far) confuses me. I followed strcpy_P definition in SAMD pgmspace.h, and it's just _#define strcpy_P(dest, src) strcpy((dest), (src))_. How can my patch (changing strcpy_P for strcpy) make it work?

It's frustrating not being able to use debug tools, I tried to import the sketch to Atmel Studio but there are a lot of compilation errors in the solution - some very weird.