TheThingsNetwork/arduino-device-lib #284: Set custom DevEUI

LasaleFamine, 2022-06-07:
Hello, I'm looking to set a custom DevEUI on my The Things Uno but I can't see any function related to this on the doc. Is there any way to do it using this library?

jpmeijers, 2022-06-07:
The RN2483 on the Things Uno has a hardware EUI that is used by this library. It's recommended to use this hardware EUI as it is guaranteed to be unique. Self assigning EUIs should only be done if the hardware does not have a unique identifier. And even when self assigning the EUI you need to use one from some other vendor's block. Normally one would generate one from the 10 free ones you get on The Things Stack CE.

So in short, don't set a custom EUI if you don't have a really good reason. The RN2483 does allow it though, but this library does not expose it. This library has been written to always use the hardware EUI as DevEUI:
https://github.com/TheThingsNetwork/arduino-device-lib/blob/d2e30f1e99aa5a1909988b0b943ee700f5c5cd2d/src/TheThingsNetwork.cpp#L664

closed 2022-06-07 by jpmeijers

LasaleFamine, 2022-06-07:
Just wondering, what if a device let's say it's broken, I want to replace it with the same device and access point on the server (so same keys), how should someone proceed? I need to register a new device on the server even if it is actually the same?