TheThingsNetwork/arduino-device-lib #290: (Question) Is this lib maintained?

LasaleFamine, 2024-10-07:
I'm wondering if this library is still maintained and if there are plans of expanding it to make it work with the newest Microchip modules. Is there anybody out there? (cit)

danalvarez, 2024-10-09:
I don't think it's mantained. I worked on my own fork that adapts part of this library to work the Microchip SAMR34/WLR089 modules. See https://github.com/danalvarez/arduino-device-lib

LasaleFamine, 2024-10-09:
That's pretty unfortunate. Great work on the fork by the way, did you also make upgrades for RN modules on the fork? Maybe we should think of some sort of community edition of this lib if TTN doesn't want to maintain it. @jpmeijers are you still on the project?

LasaleFamine, 2024-10-22:
So just to be sure, @johanstokking can you confirm this lib is unmaintained?

johanstokking, 2024-10-23:
@LasaleFamine we don't intent to make this library support other modules than the RN modules.

LasaleFamine, 2024-10-23:
@johanstokking thanks for the response. I'm gonna close the issue.

closed 2024-10-23 by LasaleFamine